memfault/memfault-firmware-sdk #10: MEMFAULT_PLATFORM_COREDUMP_STORAGE_PARTITION unused

hh-decr, 2021-04-12:
It appears that the partition NVMS_LOG_PART is hardcoded and was intended to be swapped with  MEMFAULT_PLATFORM_COREDUMP_STORAGE_PARTITION in the code but wasn't. 

See:
https://github.com/memfault/memfault-firmware-sdk/commit/1a64c90026bee493f6f92fdb760340b715c4b84d#diff-1b48dc0a21698d35d6e9d452f6a0177c4914d391f245db0d81b2d8a68794ab5aR117

https://github.com/memfault/memfault-firmware-sdk/commit/1a64c90026bee493f6f92fdb760340b715c4b84d#diff-1b48dc0a21698d35d6e9d452f6a0177c4914d391f245db0d81b2d8a68794ab5aR124

chrisc11, 2021-04-12:
Thanks for the report @hh-decr ... looks like you are right, we'll get that patched up!

chrisc11, 2021-04-20:
Fixed in release 0.16.1, thanks for the report!

closed 2021-04-20 by chrisc11