memfault/memfault-firmware-sdk #24: Is ESP32-S2 compatible with memfault

ElieSKickMaker, 2022-01-28:
Hello, 

I'm looking to use memfault on a project with a ESP32-S2. Do you know if the memfault SDK is compatible ? 

Thank you ! :)

chrisc11, 2022-02-02:
Hi @ElieSKickMaker, thanks for the report! 

To help us investigate, could you share the following:

- what version of the esp-idf you are using?
-  did you run into any issues when following the instructions in our porting guide?  https://docs.memfault.com/docs/mcu/esp32-guide/

ElieSKickMaker, 2022-02-03:
Hello, 

I'm using the esp-idf v4.3.2. 
Yes I've followed the instructions. I can build a project example ( wifi station ap + memfault port files ) like the [video ](https://www.youtube.com/watch?v=ge-ebleCi8o) if I set the target on the ESP32. But when I set the target on ESP32-S2 (idf.py set-target esp32s2) I can't build anymore. I've got some errors about undefined registers into the memfault firmware sdk..

noahp, 2022-04-08:
Hi @ElieSKickMaker , this release addresses the build errors:
https://github.com/memfault/memfault-firmware-sdk/releases/tag/0.29.0

Let us know if there's anything else causing problems!

closed 2022-04-08 by noahp