memfault/memfault-firmware-sdk #37: nrf5 SDK example on nrf52840 DK unresponsive until watchdog reset

mark-blackburn-orpyx, 2022-11-16:
I tried out this example on the nRF52840 DK:

https://github.com/memfault/memfault-firmware-sdk/tree/master/examples/nrf5/apps/memfault_demo_app

I found that it seems to hang and get unresponsive after the first run after reset. It will hang for about 10 seconds then it gets reset by the watchdog. After that it becomes responsive and the CLI will work.

Also, I was only every able to see one entry in the core storage at 0xA1000 when reading memory with the debugger, but maybe I didn't look far enough in as I don't know how much space is reserved per core dump entry.

noahp, 2022-11-17:
Hello @mark-blackburn-orpyx , thanks for posting the issue! Can you retry after flashing softdevice? in our sample app it's the `make flash_softdevice` command from here:

https://github.com/memfault/memfault-firmware-sdk/tree/0.34.2/examples/nrf5#flashing-the-demo-app

We'll check on our side too.

mark-blackburn-orpyx, 2022-11-17:
I am running the build on WSL and I don't have nrfjprog installed (and I don't frankly know how well it would work)
So instead I am loading using J-Link Commander:

```
J-Link>erase
J-Link>LoadFile C:\zip\nRF5_SDK_15.0.0_a53641a\components\softdevice\s140\hex\s140_nrf52_6.0.0_softdevice.hex
J-Link>LoadFile c:\zip\memfault-firmware-sdk\examples\nrf5\apps\memfault_demo_app\build\memfault_demo_app_nrf52840_s140.hex
J-Link>reset
J-Link>go
```

I am still seeing the same watchdog reset issue.

```
rtt_cli:~$ <info> app: MFLT: GNU Build ID: 1712d21283e2fee1111c4f1254500fef2e39bc71
<info> app: MFLT: S/N: 4115F7923D1F7641
<info> app: MFLT: SW type: nrf-main
<info> app: MFLT: SW version: 1.0.0-dev
<info> app: MFLT: HW version: pca10056
<info> app: MFLT: Reset Reason, RESETREAS=0x4
<info> app: MFLT: Reset Causes:
<info> app: MFLT:  Software
<info> app: MFLT: Memfault Initialized!
<info> app: MFLT: Software Watchdog configured to 9875ms
<info> app: main()

rtt_cli:~$  (Unresponsive)
<info> app: MFLT: GNU Build ID: 1712d21283e2fee1111c4f1254500fef2e39bc71
<info> app: MFLT: S/N: 4115F7923D1F7641
<info> app: MFLT: SW type: nrf-main
<info> app: MFLT: SW version: 1.0.0-dev
<info> app: MFLT: HW version: pca10056
<info> app: MFLT: Reset Reason, RESETREAS=0x6
<info> app: MFLT: Reset Causes:
<info> app: MFLT:  Watchdog
<info> app: MFLT: Memfault Initialized!
<info> app: MFLT: Software Watchdog configured to 9875ms
<info> app: main()
```

rtt_cli:~$ (Working now)

mark-blackburn-orpyx, 2022-11-17:
I am mixing up SDKs and softdevices. I am compiles with SDK 17.1.0 but using softdevice 6.0.0 from SDK 15.0.0

Which version of the NRF5 SDK should I be using? Which one did you test with?

mark-blackburn-orpyx, 2022-11-17:
I've answered my own question. (15.2.0) I'll have a better look at the README.md file for this example and try again.

noahp, 2022-11-18:
Correct, 15.2.0 is used in that example project. I'll test this myself tomorrow here and report back, thanks @mark-blackburn-orpyx !

mark-blackburn-orpyx, 2022-11-18:
I got it working after switching to the correct version of the SDK. This issue can be closed.

closed 2022-11-18 by mark-blackburn-orpyx

noahp, 2022-11-18:
Great, thanks for checking @mark-blackburn-orpyx , much appreciate.