memfault/memfault-firmware-sdk #48: memfault is `select`ing a non-existent (anymore) Kconfig option

carlescufi, 2023-04-04:
[This line](https://github.com/memfault/memfault-firmware-sdk/blob/0d7e77332e16279773ee59cb62c2f6be60816d78/ports/zephyr/Kconfig#L8) selects a Kconfig option, `LEGACY_INCLUDE_PATH`, that was removed after the release of Zephyr v3.3.0: https://github.com/zephyrproject-rtos/zephyr/commit/9b30667c7720029f68adc2844e81938a76a67c2a.

There is a workaround implemented in memfault's [CMakeLists.txt](https://github.com/memfault/memfault-firmware-sdk/blob/0d7e77332e16279773ee59cb62c2f6be60816d78/ports/zephyr/CMakeLists.txt#L63) that should make the `select` statement redundant.

noahp, 2023-04-04:
Thanks for filing @carlescufi , we'll fix this asap!

noahp, 2023-04-05:
FYI this is fixed in [0.42.1](https://github.com/memfault/memfault-firmware-sdk/releases/tag/0.42.1), thanks for filing this issue!

closed 2023-04-05 by noahp