memfault/memfault-firmware-sdk #49: zephyr: z_NmiHandlerSet used in port is gone

mbolivar-nordic, 2023-04-18:
Zephyr commit https://github.com/zephyrproject-rtos/zephyr/commit/77d2490164cbb5dbdd8dee9c0787204b0006f635 renamed this function.

This is breaking the build at least in ports/zephyr/v2.4/memfault_fault_handler.c:

```
static int prv_install_nmi_handler() {
  extern void z_NmiHandlerSet(void (*pHandler)(void));
  z_NmiHandlerSet(MEMFAULT_EXC_HANDLER_NMI);
  return 0;
}
```

I'm not sure what the right fix is -- perhaps a new ports/zephyr/v3.4 which uses the new name, `z_arm_nmi_set_handler`?

mbolivar-nordic, 2023-04-18:
This is affecting the nRF Connect SDK.

noahp, 2023-04-26:
We've just shipped a release with a fix for this, thank you for posting!
https://github.com/memfault/memfault-firmware-sdk/releases/tag/0.43.1

closed 2023-04-26 by noahp