memfault/memfault-firmware-sdk #52: zephyr: Separate connect() from prv_configure_socket()

anicare-tero, 2023-06-16:
Hi,

I'm using nRF9160 on nRF Connect SDK version 2.3.0. We have had some problems with Memfault cloud where the connecting or data sending is taking too much time and the watchdog kicks in and reboots the device.

Could the `connect()` function be separated from the `prv_configure_socket()` function? So when the socket would be opened first by calling `memfault_zephyr_port_http_open_socket()`, then I would be able to setup socket timeouts and then call the new function `memfault_zephyr_port_http_connect()`.

I have some kind of proposal about this:

ports/zephyr/common/memfault_platform_http.c
```
static int prv_configure_socket(int fd, const char *host) {
...
  return 0;
}

int memfault_zephyr_port_http_connect(sMemfaultHttpContext *ctx) {
  if (ctx->sock_fd < 0) {
    memfault_zephyr_port_http_close_socket(ctx);
    return -1;
  }

  return connect(ctx->sock_fd, ctx->res->ai_addr, ctx->res->ai_addrlen);
}
```

I think this would require some other code changes too but this would be really nice to have.

ejohnso49, 2023-06-16:
Thank you for submitting this issue! For my background, could you tell me a little bit about the length of the timeouts you would enable and the watchdog period you have configured?

ejohnso49, 2023-07-13:
Just wanted to leave a quick update, this work is in progress! Thanks!

ejohnso49, 2023-08-30:
@anicare-tero Apologies for the delay, but we shipped this in 1.1.3 on 2023-08-08. Please give it a try and feel free to open an issue for any further improvements. Thanks!

closed 2023-08-30 by ejohnso49