memfault/memfault-firmware-sdk #57: zephyr: legacy include workaround

JordanYates, 2023-07-05:
The legacy workaround kconfig (`CONFIG_LEGACY_INCLUDE_PATH`) has been removed with v3.4 (https://github.com/zephyrproject-rtos/zephyr/commit/9b30667c7720029f68adc2844e81938a76a67c2a).

The Memfault SDK is globally undoing that change with the following lines:
https://github.com/memfault/memfault-firmware-sdk/blob/73ec7b7c020d9183f264c45c0fa3d3f2b9238aa1/ports/zephyr/CMakeLists.txt#L57-L65

ejohnso49, 2023-07-05:
We use this as a way to simplify our code and maintain compatibility with older versions of Zephyr and our SDK. Are you hitting building warnings on v3.4? Would you mind pasting any output that you hit? I have a build locally on 3.4 but I do not see any warnings regarding the include path.

JordanYates, 2023-07-05:
It is not creating warnings, but it is messing with intellisense, as well as negating the benefits of https://github.com/zephyrproject-rtos/zephyr/issues/41543

ejohnso49, 2023-08-31:
We just released a new SDK version, 1.2.0, which should address this issue. Thanks again for reporting

closed 2023-08-31 by ejohnso49