memfault/memfault-firmware-sdk #66: rand32.h needs changing to random.h for zephyr 3.5

nordicjm, 2023-11-17:
Zephyr 3.5 deprecated rand32.h, the new include is random.h, this needs to be updated in `ports/zephyr/common/memfault_http_periodic_upload.c`

chrisc11, 2023-11-23:
Thanks for the report @nordicjm -- we'll have the update published in our release next week

nordicjm, 2023-11-29:
@chrisc11 just checking which day you expect this release to be published on github?

noahp, 2023-11-30:
@nordicjm - We hope to have this release (1.5.0) published tomorrow, 2023-11-30.

nordicjm, 2023-12-07:
This appears to be a broken release, there is this unconditional cmake code in `ports/zephyr/ncs/CMakeLists.txt`:
```
  if (NCS_VERSION_MAJOR)
    # Note: Starting in nRF Connect SDK >= 1.3, NCS_VERSION_* fields are exposed
    if (${NCS_VERSION_MAJOR}.${NCS_VERSION_MINOR}.${NCS_VERSION_PATCH} GREATER_EQUAL 2.4.0)
      # We wrap download_client_get() in order to register multiple root certificates
      #   See comment in src/memfault_fota.c for more details
      zephyr_ld_options(-Wl,--wrap=download_client_get)
    endif()
  endif()
```
Which then causes a compilation failure because also in this file, the .c file that contains these wrapped functions is only included if a Kconfig is set:
```
zephyr_library_sources_ifdef(CONFIG_MEMFAULT_FOTA memfault_fota.c)
```
So when the download client is enabled and memfault FOTA is disabled, this prevents compilation of projects

noahp, 2023-12-07:
@nordicjm Memfault SDK v1.5.1 is now available, with the fix for this issue

nordicjm, 2023-12-08:
Perfect, all working, many thanks @noahp!

closed 2023-12-08 by nordicjm