memfault/memfault-firmware-sdk #68: Cannot use external RTC for Zephyr projects

corytodd, 2024-04-25:
For Zephyr ports, Memfault expects the RTC device to be node named rtc. e.g.

```
rtc: rtc_foo@address {
    compatible = "vendor,rtc_foo";
    reg = <address>;
    status = "okay";
};
```

When using an external RTC we will encounter issues in the base address validation on nrf parts, e.g.
```
zephyr/soc/arm/nordic_nrf/validate_base_addresses.c:218:19: error: 'NRF_RTC' undeclared here (not in a function); did you mean 'NRF_RTC0'?
  218 | CHECK_DT_REG(rtc, NRF_RTC);
```

I propose that we add a fallback to allow using an alias if the node is not found. This should provide backwards
compatibility and support external RTCs simultaneously.

---

Zephyr-Sdk: 0.16.5
nrfconnect: v2.6.0
Memfault SDK: 1.6.0

noahp, 2024-04-26:
Thank you! I had tried and failed to figure out the error I had in this implementation. We'll get your patch included in our next release :+1:

ejohnso49, 2024-06-28:
Hi @corytodd sorry for the delayed response. We've incorporated this into the SDK starting with version 1.9.0

closed 2024-06-28 by ejohnso49