memfault/memfault-firmware-sdk #72: Missing 'PRIu64' definition

iotengtr, 2024-07-23:
I get undefined PRIu64 errors like the following sample in the `memfault_self_test.c` file: 
```
memfault-src/components/core/src/memfault_self_test.c:339:80: error: expected ')' before 'PRIu64'
  339 |     MEMFAULT_LOG_ERROR("Time since boot not monotonically increasing: start[%" PRIu64

```
The problem can be solved by adding manually PRIu64 definition macro.

The platform info: 
`Ubuntu 22.04.4 LTS`
`arm-none-eabi-g++ (15:10.3-2021.07-4) 10.3.1 20210621 (release)`
```
cmake version 3.29.6
CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

noahp, 2024-07-23:
@iotengtr thank you for reporting the issue! we'll make a change to fix the problem.

iotengtr, 2024-07-23:
> @iotengtr thank you for reporting the issue! we'll make a change to fix the problem.

Thanks @noahp for quick response. Success.

noahp, 2024-07-24:
@iotengtr FYI the fix for this has been published today as part of SDK release v1.10.1. Closing for now, please re-open if there is additional problems!

closed 2024-07-24 by noahp