memfault/memfault-firmware-sdk #77: `memfault_platform_coredump_save_begin` is documented as being weak no-op, but is not in ESP-IDF

finger563, 2024-10-21:
Per the [documentation](https://docs.memfault.com/docs/mcu/coredump-sensitive-data#wiping-sensitive-data-before-collection) and within the [base sdk](https://github.com/memfault/memfault-firmware-sdk/blob/master/components/panics/src/memfault_coredump.c#L528) `memfault_platform_coredump_save_begin()` is listed as a weakly defined symbol that can be overridden by user code.

However, the `esp-idf` platform port overrides this [here](https://github.com/memfault/memfault-firmware-sdk/blob/master/ports/esp_idf/memfault/common/memfault_platform_coredump.c#L365).

noahp, 2024-10-21:
Hi @finger563 , you're correct! We'll update the SDK to make that change. Do you have a timeline for when you'd need an SDK release with the fix?

finger563, 2024-10-21:
We have some platform integration deliverable timelines which are due by Nov 11, though of course the earlier the better for completing the integration and testing.

noahp, 2024-10-21:
Thanks @finger563 ! We'll schedule an SDK release for this week, I'll post back here when it's shipped.

gminn, 2024-10-24:
@finger563 This has been addressed in v1.16.0, the release that went out today. Please review the Changelog [here](https://github.com/memfault/memfault-firmware-sdk/blob/master/CHANGELOG.md#-added) for details, and let us know if you have any other questions!

closed 2024-10-24 by gminn