memfault/memfault-firmware-sdk #81: `v1.19.0`: Zephyr Kconfig not being included

JordanYates, 2025-01-11:
The change to set `kconfig-ext` in the latest release means that none of the Kconfig options in `ports/zephyr` are included in the build.
https://github.com/memfault/memfault-firmware-sdk/blob/03050f22d52c4930172a9707e30740aaf4c2836d/zephyr/module.yml#L7

`kconfig-ext` skips evaluation of any further module kconfig settings:
https://github.com/zephyrproject-rtos/zephyr/blob/1f4874088f17c2d617426aa8cb78b66bbb634989/scripts/zephyr_module.py#L384-L392

This results in build failures when attempting to build after upgrading from `v1.18.0`:
```
/home/jordan/code/workspace/zephyr/scripts/kconfig/kconfig.py: /home/jordan/code/workspace/build/Kconfig/Kconfig.modules:14: Could not open '/home/jordan/code/workspace/zephyr/' (in 'osource "$(ZEPHYR_MEMFAULT_KCONFIG)"') (EISDIR: Is a directory)
```

Reverting the change to `kconfig: ports/zephyr/Kconfig` fixes builds.
This change was not mentioned in the changelog.

noahp, 2025-01-21:
Hi @JordanYates - Thanks for reporting this problem! Can you provide a little more detail about the issue- which version of Zephyr are you using? Do you have an example that I can use to reproduce the problem?

Our in-tree sample app here doesn't appear to be affected by the problem (but I may be missing something obvious):
https://github.com/memfault/memfault-firmware-sdk/tree/master/examples/zephyr/qemu

JordanYates, 2025-01-21:
```
(.zephyr_venv) jordan@TAURUS:~/code/zephyr/zephyr$ git describe
v3.7.0
(.zephyr_venv) jordan@TAURUS:~/code/zephyr/zephyr$ git diff
diff --git a/west.yml b/west.yml
index 9454f2e57ad..d2d0bf865e7 100644
--- a/west.yml
+++ b/west.yml
@@ -23,6 +23,8 @@ manifest:
       url-base: https://github.com/zephyrproject-rtos
     - name: babblesim
       url-base: https://github.com/BabbleSim
+    - name: memfault
+      url-base: https://github.com/memfault

   group-filter: [-babblesim, -optional]

@@ -342,6 +344,10 @@ manifest:
     - name: zcbor
       revision: 75d088037eb237b18e7ec1f47c9ce494b9b95aab
       path: modules/lib/zcbor
+    - name: memfault-firmware-sdk
+      path: modules/lib/memfault
+      revision: 1.19.0
+      remote: memfault

   self:
     path: zephyr
(.zephyr_venv) jordan@TAURUS:~/code/zephyr/zephyr$ west update
...
(.zephyr_venv) jordan@TAURUS:~/code/zephyr/zephyr$ west build -p -b nrf52840dk/nrf52840 samples/hello_world/ -- -DCONFIG_MEMFAULT=y
...
/home/jordan/code/zephyr/zephyr/scripts/kconfig/kconfig.py: build/Kconfig/Kconfig.modules:242: Could not open '/home/jordan/code/zephyr/zephyr/' (in 'osource "$(ZEPHYR_MEMFAULT_KCONFIG)"') (EISDIR: Is a directory)
CMake Error at /home/jordan/code/zephyr/zephyr/cmake/modules/kconfig.cmake:389 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/jordan/code/zephyr/zephyr/cmake/modules/zephyr_default.cmake:132 (include)
  /home/jordan/code/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/jordan/code/zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/home/jordan/code/.zephyr_venv/bin/python3 -B/home/jordan/code/zephyr/zephyr/build -GNinja -DBOARD=nrf52840dk/nrf52840 -DCONFIG_MEMFAULT=y -S/home/jordan/code/zephyr/zephyr/samples/hello_world
```
The exact same error occurs on `v4.0.0` as well.

The purpose of `kconfig-ext` and the other options is when you want the build system integration (cmake, kconfig, etc) external to the repository hosting the code (see https://docs.zephyrproject.org/latest/develop/modules.html#module-integration-files-external)

This makes sense for HAL layers or w/e that exist as repos from manufacturers that Zephyr doesn't want to touch if it can help it. It doesn't make sense for  Memfault when you're just trying to point back to your own repo anyway.

noahp, 2025-01-22:
Thank you for the sample! I tested in a clean zephyr workspace:

```bash
❯ mkdir zephyr-3.7.0
❯ cd zephyr-3.7.0
❯ west init -o=--depth=1 --manifest-url https://github.com/zephyrproject-rtos/zephyr.git --manifest-rev v3.7.0
❯ echo 'diff --git a/west.yml b/west.yml
index 9454f2e57ad..b1675e3b3d3 100644
--- a/west.yml
+++ b/west.yml
@@ -343,6 +343,11 @@ manifest:
       revision: 75d088037eb237b18e7ec1f47c9ce494b9b95aab
       path: modules/lib/zcbor

+    - name: memfault-firmware-sdk
+      url: https://github.com/memfault/memfault-firmware-sdk
+      path: modules/lib/memfault-firmware-sdk
+      revision: 1.19.0
+
   self:
     path: zephyr
     west-commands: scripts/west-commands.yml
' | git -C zephyr apply

❯ git --no-pager -C zephyr diff
diff --git a/west.yml b/west.yml
index 9454f2e5..b1675e3b 100644
--- a/west.yml
+++ b/west.yml
@@ -343,6 +343,11 @@ manifest:
       revision: 75d088037eb237b18e7ec1f47c9ce494b9b95aab
       path: modules/lib/zcbor

+    - name: memfault-firmware-sdk
+      url: https://github.com/memfault/memfault-firmware-sdk
+      path: modules/lib/memfault-firmware-sdk
+      revision: 1.19.0

❯ west update --narrow --fetch-opt=--depth=1
❯ west build -p -b qemu_cortex_m3 zephyr/samples/hello_world/ -- -DCONFIG_MEMFAULT=y
...
[173/173] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       20412 B       256 KB      7.79%
             RAM:       16104 B        64 KB     24.57%
        IDT_LIST:          0 GB        32 KB      0.00%
Generating files from /home/noah/dev/memfault/zephyrs/zephyr-3.7.0/build/zephyr/zephyr.elf for board: qemu_cortex_m3
```

Is it possible your zephyr workspace has some additional modules that are somehow interfering with the build? Your example uses the `nrf52840dk/nrf52840` board, which in a pure zephyr (not nrf-connect-sdk) workspace, Memfault will emit a build error (some additional features are implemented in the nrf-connect-sdk for certain SOCs, and is enabled by default in the Memfault SDK here: https://github.com/memfault/memfault-firmware-sdk/blob/03050f22d52c4930172a9707e30740aaf4c2836d/ports/zephyr/ncs/Kconfig#L3 ):

```bash
CMake Error at /home/noah/dev/memfault/zephyrs/zephyr-3.7.0/modules/lib/memfault-firmware-sdk/ports/zephyr/ncs/CMakeLists.txt:4 (message):
  Memfault SDK requires nRF Connect SDK version 1.9.2 or newer.  Please
  contact mflt.io/contact-support for assistance.
```

The reason we made the change- it's a bit of a workaround to conditionally include Kconfig fragments on a per-Zephyr-version basis (see suggestion here https://github.com/memfault/memfault-firmware-sdk/pull/26 ). Previously we used cmake to provide zephyr-version-specific workarounds, which was far from ideal 😕

noahp, 2025-01-22:
Ah, it looks like this snippet is relying on the module directory being `memfault-firmware-sdk`, but you are specifying it as `memfault`:

```yaml
path: modules/lib/memfault
```

noahp, 2025-01-22:
I think the Memfault SDK is missing this, to handle the case you ran into:

```bash
❯ git --no-pager diff -- zephyr/module.yml
diff --git a/zephyr/module.yml b/zephyr/module.yml
index ee3282a..ef78690 100644
--- a/zephyr/module.yml
+++ b/zephyr/module.yml
@@ -1,3 +1,5 @@
+name: memfault-firmware-sdk
+
 build:
   cmake: ports/zephyr
   # Enable external module integration files. This enables customizing the
```

JordanYates, 2025-01-23:
> The reason we made the change- it's a bit of a workaround to conditionally include Kconfig fragments on a per-Zephyr-version basis (see suggestion here [...]

I don't understand how this helps resolve any issues. You're still pointing at a file like `kconfig: ports/zephyr/Kconfig` used to, but adding an additional requirement that the modules live at an expected path (which is by no means guaranteed, since the `path` key in manifests exists so that users can change the path as they please).

noahp, 2025-01-24:
Apologies, I should have posted some more context:

The problem is we are setting `ZEPHYR_MEMFAULT_FIRMWARE_SDK_KCONFIG` here explicitly:
https://github.com/memfault/memfault-firmware-sdk/blob/03050f22d52c4930172a9707e30740aaf4c2836d/ports/zephyr/modules/modules.cmake#L10

But the `<name>` segment of that variable is taken from the module `name` property, defaulting to the path:

https://github.com/zephyrproject-rtos/zephyr/blob/36940db938a8f4a1e919496793ed439850a221c2/scripts/zephyr_module.py#L206

If we set it explicitly in `module.yaml`, it works as intended. I've also confirmed the name specified in `west.yml` is not used during this stage of module processing, so it works with a manifest like this, for example:

```yaml
    # both module name and path are NOT 'memfault-firmware-sdk'
    - name: memfault
      url: https://github.com/memfault/memfault-firmware-sdk
      path: modules/lib/memfault
      revision: 1.19.0
```

Please let me know if the change works for you!