memfault/memfault-firmware-sdk #82: `MEMFAULT_METRICS_THREADS`: missing zephyr dependencies

JordanYates, 2025-01-11:
The new `MEMFAULT_METRICS_THREADS` option only has `depends on THREAD_STACK_INFO`, but it unconditionally calls `k_thread_stack_space_get` in `prv_get_stack_usage_pct`. `z_impl_k_thread_stack_space_get` is only defined when both `INIT_STACKS` and `THREAD_STACK_INFO` are defined.

https://github.com/zephyrproject-rtos/zephyr/blob/1f4874088f17c2d617426aa8cb78b66bbb634989/include/zephyr/kernel.h#L498

Leads to compiler errors after upgrading to `v1.19.0`.
```
modules/memfault/lib..__modules__lib__memfault__ports__zephyr.a(memfault_platform_thread_metrics.c.obj): in function `k_thread_stack_space_get':
/home/jordan/code/workspace/build/zephyr/include/generated/zephyr/syscalls/kernel.h:102: undefined reference to `z_impl_k_thread_stack_space_get'
```

noahp, 2025-01-13:
Thanks @JordanYates , great catch! We'll make the fix in the next SDK release.

noahp, 2025-02-10:
This is fixed in [1.20.0](https://github.com/memfault/memfault-firmware-sdk/releases/tag/1.20.0). Let me know if there's any issues!

closed 2025-02-10 by noahp