memfault/memfault-firmware-sdk #86: feature request: Option to change the network timeout for Zephyr

chirambaht, 2025-01-14:
One of the key function used when collecting data from the Memfault servers is `prv_poll_socket`. In this function, a constant, pre-defined value of 5000ms is used for the network connection timeout. In the ESP-IDF port, I saw that this has already been done for some of the other networking functions using `CONFIG_MEMFAULT_HTTP_CLIENT_TIMEOUT_MS`. It would be nice to also have this value be defined by a Kconfig in much the same way for the Zephyr port as well.

noahp, 2025-01-14:
@chirambaht thanks for reporting! great idea, we will add it to the next SDK release. Let me know if there's urgency for a new SDK with the changes, and we can expedite it- otherwise, it's scheduled for sometime next week.

noahp, 2025-02-10:
This is fixed in [1.20.0](https://github.com/memfault/memfault-firmware-sdk/releases/tag/1.20.0). Let me know if there's any issues!

closed 2025-02-10 by noahp