memfault/memfault-firmware-sdk #87: iar build failure due to enum MEMFAULT_PACKED

zejiang0jason, 2025-01-24:
Hello,

I got such compile error with iar
iar: 9.60.3
memfault-sdk: 1.8.0

error:
> 	Error[Pe110]: expected either a definition or a tag name	memfault-firmware-sdk\components\include\memfault\core\reboot_reason_types.h	29
>	Error[Pe040]: expected an identifier	memfault-firmware-sdk\components\include\memfault\core\reboot_reason_types.h	29
>	Error[Pe065]: expected a ";"	memfault-firmware-sdk\components\include\memfault\core\reboot_reason_types.h	29

It is because of the `MEMFAULT_PACKED` used in:
https://github.com/memfault/memfault-firmware-sdk/blob/03050f22d52c4930172a9707e30740aaf4c2836d/components/include/memfault/core/reboot_reason_types.h#L29

I am not sure whether it is similar to: https://github.com/IntergatedCircuits/USBDevice/issues/14. But when I changed to `typedef MEMFAULT_PACKED enum MfltResetReason`, IAR build pass.

Could you please help? Thank you very mch.

noahp, 2025-01-27:
Hi @zejiang0jason , thanks for reporting! We are in progress fixing this issue now, we will include the fix in our next SDK release (scheduled for this week). Please let me know if you need a release sooner, and we will expedite it!

zejiang0jason, 2025-02-05:
Hi @noahp , thanks for feedback, when will the fix be included? this week? Yes, our SW release depends on this fix. If possible, please expedite it, thank you very much.

zejiang0jason, 2025-02-12:
thanks, it works.