memfault/memfault-firmware-sdk #89: `MEMFAULT_NRF_CONNECT_SDK` enabled by default in upstream Zephyr

JordanYates, 2025-02-03:
The `MEMFAULT_NRF_CONNECT_SDK` is currently selected based only on the enabled SoC.
`default y if SOC_SERIES_NRF52X || SOC_SERIES_NRF53X || SOC_SERIES_NRF91X`
As a result, the symbol ends up enabled when Memfault is imported from an upstream project, even though NCS doesn't exist anywhere.

This default could be extended to also require the autogenerated module Kconfig symbol for the NRF repositories to be enabled. For example:
`default y if ZEPHYR_NRF_MODULE && (SOC_SERIES_NRF52X || SOC_SERIES_NRF53X || SOC_SERIES_NRF91X)`

This would preserve the default for NCS users, while preventing it from being enabled by default for upstream users.

noahp, 2025-02-04:
Excellent idea, thanks @JordanYates ! We'll include this in the next SDK release.

noahp, 2025-02-10:
This is fixed in [1.20.0](https://github.com/memfault/memfault-firmware-sdk/releases/tag/1.20.0) (apologies, did not include a changelog entry, I'll fix that!). Let me know if there's any issues!